IRNAS/PiRa-zero #2: Charger not working, fault condition

SloMusti, 2017-10-07:
The bq24296 charger LED is blinking with 1Hz and reports a fault. Charger is not working. Evaluating the conditions that could lead to this:

The REGN is enabled when all the conditions are valid.
1. VBUS above VVBUS_UVLOZ  (3.6V) - measured 5.1V
2. VBUS above VBAT+ VSLEEPZ in buck mode or VBUS below VBAT+ VSLEEPin boost mode. Vbat 3.2V, correct.
3. After typical 220-ms delay (100 ms minimum) is complete

If one of the above conditions is not valid, the device is in high impedance mode (HIZ) with REGN LDO off. 

Given this is not the cause, there must be some correlation to HIZ mode. PG is low, so input voltage check is completed. TS circuit is checked and operates as expected.

Checking I2C registers for faults:

USB disconnected:
```
pi@ ~ $ i2cdump -r 0x00-0x0a 1 0x6b
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 37 1b 60 11 b2 9c 73 4b 00 80 20                   7?`???sK.?
```

USB connected:
```
pi@ ~ $ i2cdump -r 0x00-0x0a 1 0x6b
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 37 1b 60 11 b2 9c 73 4b 85 81 20                   7?`???sK??
```

SloMusti, 2017-10-07:
Timer and NTC faults are indicated. REGN must work for NTC fault to be cleared, so that can not happen.

VojislavM, 2017-11-02:
bad soldering problem

closed 2017-11-02 by VojislavM